KiCad/kicad-library-utils #11: invalid Violating Rule 6.5 message

pointhi, 2015-12-12:
it seems like the ```check_kicad_mod.py``` has a little bug with the orientation of square pads, because it throws an invalid error for some of my footprints, where no 6.5 violation is visible.

Furthermore, when using ```--fix``` it delets silkscreen, which would be overlapping the pad, when the pad would be orientated 90 degrees inclined, but not how I have inclined it in the footprint.

### Footprint before --fix (purporsed violation of rule 6.5)

[image]

## Footprint after --fix
[image]

Footprints can be found in this PR: https://github.com/KiCad/RF_Modules.pretty/pull/1

ricardocrudo, 2015-12-14:
@andres53016 can you check it, please?

andres53016, 2015-12-14:
Ok! I fix the bug and I create new PR

twam, 2016-01-09:
Encountered the same bug here, the provided bug fix solves the issue.

ricardocrudo, 2016-01-19:
Fixed.

pointhi, 2016-02-02:
Hm, I don't know why, but now many other footprints throw 6.5 Errors for me.

For example the MC9A12 footprints: https://github.com/KiCad/Connectors_Multicomp.pretty/pull/1):
```
checking module: Multicomp_MC9A12_02x32x2.54mm_Straight_MC9A12-6434
  Violating Rule 6.5
    Silkscreen is not superposed to pads, its outline is completely visible after board assembly, uses
    0.15mm line width and provides a reference mark for pin 1. (IPC-7351).
```

The old script doesn't throw errors for those footprints.

ricardocrudo, 2016-03-06:
@andres53016  Can you check that when you got some spare time?

andres53016, 2016-03-06:
Hi @ricardocrudo  
It was right on that now.

CarlPoirier, 2016-03-06:
New PR is https://github.com/KiCad/kicad-library-utils/pull/20

CarlPoirier, 2016-03-07:
I merged the fix for the rule 6.5 false positive.

closed 2016-03-07 by CarlPoirier