KiCad/kicad-library-utils #17: [Enhancement] checklib structure

diggit, 2016-03-05:
I was browsing sources and noticed, that verbose messages are generated in checklib.py script. This is kind of limiting, because each rule script returns just T/F. All additional info is specific for each rule and checklib.py must handle message for each rule separately.
I suggest:
- add *verbose level argument* into rule constructor (rule.py)
- add *list of messages* which will buffer verbose message from rule script (rule.py)
- add method for appending separate verbose messages into *message list* ^^ (rule.py)
- modify all rule scripts to compose verbose messages using that method^^
- modify checklib.py to pass verbose level to rule scripts and print message buffers after check() (this will be unified for all rules, so there will be no need to modify checklib.py for each rule script)

I can do this and give you PR. I just need to know, if you agree with this idea.

ricardocrudo, 2016-03-05:
Yeah, great idea.

diggit, 2016-03-05:
It is almost done, but I got stuck on:
https://github.com/KiCad/kicad-library-utils/blob/master/schlib/rules/EC01.py#L28
What is this checking? Tilde on both sides means that pin should be POWER INPUT?

ricardocrudo, 2016-03-05:
Sorry, really poor comment there.
That's to check if pin name got a line over it, meaning inverted function. When pin name already has this line to indicate it's inverted, thus the pin draw cannot has the circle which also indicate inverted function. In short, pin must have only one indication of inverted function, by having a circle in the end of the pin or having a line over its name.

diggit, 2016-03-05:
Oh, I see. Then setting pin type to W in fix() is not the right solution. It will suppress error next time, but changing INPUT -> POWER INPUT is not the best way how to "fix" it.
And second point, KiCad allows non closed overlines like ~NAME. Used regex won't match these. Just letting you know...

Should be preserved dot or overline?

ricardocrudo, 2016-03-06:
> Oh, I see. Then setting pin type to W in fix() is not the right solution. It will suppress error next time, but changing INPUT -> POWER INPUT is not the best way how to "fix" it.

Indeed not, that's wrong! I added this check later and I forgot about the fix part.

> And second point, KiCad allows non closed overlines like ~NAME. Used regex won't match these. Just letting you know...

Good to know

> Should be preserved dot or overline?

I've seen more overline, so I would go for it

PR #19 (2016-03-06): Improve verbose output
Improved way how to generate verbose output.
Now each rule script fills its own message buffer with success/info/warning/error messages. checklib.py then selects which to print and how to colorize them. (each message has verbosity and severity flag)
All scripts have message for each detected problem. So no more guessing what could be wrong.

Scripts were tested on KiCad library.

These changes makes a lot of variables unused. Then could be removed in future.

This PR resolves #17
closed 2016-03-06 by ricardocrudo