KiCad/kicad-library-utils #30: move_part.py does not move alias documentation

diggit, 2016-07-04:
When moving parts with script `move_part.py` symbol and documentation of main symbol is moved. Unfortunately documentation of all aliases remain in source dcm file.

Once this is fixed, checklib.py should be extended to check unused aliases. There could be a lot of them.

diggit, 2016-07-04:
@ricardocrudo I noticed, that dcm file write operations are not handled by Schlib class and have to be done manually. What do you think about integration of this functionality into it? This could solve problems (simplify solution) with move_part.py and simplify future use.

ricardocrudo, 2016-07-05:
Hello @diggit, this issue is duplicated with #4 

I think it is a good idea to integrate them.

diggit, 2016-07-05:
Dang, 3 opened issues and I create duplicate. I'll take care of this.

PR #32 (2016-07-10): Fix schlib.py andmove_part.py to move components properly
I made some changes to be able to move components properly without abandoning aliases documentation.

SchLib class nnow has 2 additional methods:
- addComponent( component_reference )
- removeComponent( component_name )
 - returns reference to removed component

Both methods handles symbol + documentation (incl. aliases).

This PR fixes #30 and #4
closed 2016-07-10 by ricardocrudo