KiCad/kicad-library-utils #37: New EC for duplicate graphic elements

diggit, 2016-07-10:
It can easily happen when you are copying same parts of symbol.
- SIMPLE - check for g. elements with same points

or
- COMPLEX - find all overlapping lines no matter of what are they part of (polyline, rectangle), no matter of their length

What do you think?

SchrodingersGat, 2016-07-10:
I think that the SIMPLE options is a good start, and would be worlds easier to implement.

If you have the time and inclination to implement the COMPLEX option, then go for it, although I can see that raising a lot more false-positives than the SIMPLE one.

cpresser, 2019-05-05:
This looks like a similar challenge than https://github.com/KiCad/kicad-library-utils/pull/286 
I think that the code in there can be easily adopted to work for symbols as well.
Is anyone aware of a (real) symbol with this problem that I can use for testing?

evanshultz, 2019-05-06:
I recall seeing a few PRs I merged recently with overlapping or duplicate stuff. I don't recall all the details but I searched a few keywords and they may help:
- https://github.com/KiCad/kicad-symbols/pull/1683
- https://github.com/KiCad/kicad-symbols/pull/668

I could make up a test case if you're looking for one. What did you have in mind to cover?

Note that a polyline and rectangle could overlap. Or the points in a polyline could be in reverse order. Perhaps other geometries can cause duplication of elements with no visual change as well. So doing this thoroughly won't be the same as checking for duplicate lines.