KiCad/kicad-library-utils #47: Update checklib.py script for automated testing

reportingsjr, 2016-08-25:
I would love to get the automated testing merged in to master so I don't have to worry about them diverging as much. I'll use this as a starting point to figure out a good way to do this. Feel free to chime in if you feel like something should change

The name of the game is to /only/ check symbols (footprints will come later) that have been changed in a pull request.

Diffs often don't include the full symbol source that has been changed which is annoying, but understandable. So we have to use some other way to figure out exactly which symbols were changed.

The way I have it set up to do this is by running checklib.py on the master, storing the output of that in a file, then running checklib.py on the pull request branch, and lastly I compare the output of the two. As long as no new errors were added I consider it a successful pull request and update the status.

It would be nice if instead it could just detect if a symbol had been changed and mark the pull request as failed if there are any errors on that symbol, whether or not the errors on it are new. It would likely be possible to do this in a similar way to how the errors are done now, where we parse all of the symbols on master and store those, then parse all of the symbols on the pr branch, and then compare. Not sure how much work this would require though.

Minor point: this will likely get moved to a different script called automatedtesting.py so checklib.py doesn't end up super cluttered.

The current setup can be seen on my kicad-library-utils mirror (http://github.com/reportingsjr/kicad-library-utils/)

SchrodingersGat, 2016-08-25:
I had an idea for detecting whocu symbols have changed. If we create some sort of checksum algorithm for each symbol, we can parse the master copy, and generate a name:checksum pair for each symbol.

Then run the same check on the current branch and the changed symbols are the ones with a checksum mismatch. 

What do you think?

SchrodingersGat, 2016-08-27:
Jon, I have implemented component-level checksum and added a python script for running KLC on any components that have been added / updated into a .lib file.

Now we only need to look at which .lib files have been touched, and pass the old/new versions to the script.

https://github.com/KiCad/kicad-library-utils/pull/48

reportingsjr, 2016-08-27:
Well that was fast!

It looks like a pretty reasonable way to do this really. It looks like it needs a bit of adjusting before it will be fully suitable for jenkins, but that is just small tweaks about printing output.

I might pull that in to my own branch before I would suggest merging it in to this repo so I can get everything tidied up. I either need someone with admin access to respond  to me, or I need to wait a few more days until I can put my server back online.

I'll keep you posted.

SchrodingersGat, 2016-08-29:
One addition that is needed to the script is to give proper return codes based on the KLC checks - currently the script doesn't return anything

CarlPoirier, 2016-08-31:
#48 is merged :+1:

reportingsjr, 2016-09-07:
This is in a much better state now so I'll go ahead and close it. Thanks you two! Fantastic work!

closed 2016-09-07 by reportingsjr