KiCad/kicad-library-utils #57: check_kicad_mod.py does not shift 3D models when shifting the origin!!!

jkriege2, 2016-10-01:


diggit, 2016-10-01:
This is problem of 3D models with **wrong origin placement**. (It does not make sense to create 3D models for non KLC compliant footprints.)

I don't see this as error.

Ratfink, 2018-04-18:
Could we try to get a decision on this issue?  As it is, it's unclear if the current behavior is a problem or not.  Overall I agree with what Patrik said, but other people may still disagree, and without any discussion it's hard to know what to do.

evanshultz, 2018-04-18:
It's unclear to me as well. If this request is to move the 3D model when fixing the footprint, I think that's a 3D model problem and shouldn't be touched by this script. It seems too fraught with potential gotchas to move a 3D model origin if the footprint is getting shifted. Maybe the 3D model and footprint aren't aligned and so an equivalent shift in both doesn't align them! But maybe there's more to it.