KiCad/kicad-library-utils #62: Rule 6.5 violation even thought no overlap existst (and other silk features are at same distance to pad)

poeschlr, 2016-10-18:
Hello, i found a curious bug in the rule 6.5 script.

It tells me that two of my silkscreen lines violate the rule even if i move them farther away than others.
Testfiles: (one with only the violating lines on f.silk the other one with all non violating lines.)
[Testfiles.pretty.zip](https://github.com/KiCad/kicad-library-utils/files/535592/Testfiles.pretty.zip)

Commit hash with the problem: e72a61a

ricardocrudo, 2016-11-19:
hello @poeschlr I've tested your files against the last version of check_kicad_mod.py script and I didn't get any error. Can you please confirm the problem is solved?

poeschlr, 2016-11-24:
Hello @reportingsjr,
the problem still exists. (Commit hash ca9e655)
Python 2.7.11 (default, Sep 29 2016, 13:33:00) 
[GCC 5.3.1 20160406 (Red Hat 5.3.1-6)] on linux2

The problem seems to be some rounding errors.
If i change all divisions by 2 to divisions by 2.0 the problem is solved.
(Pad['size'][?] is not guaranteed to be a floating point number. Python is strange sometimes.)

ricardocrudo, 2016-11-24:
Hi @poeschlr, indeed it's an issue about python 2 and 3. Thanks for debug it. Although your pr #67 fix the problem I'd recommend to use the following `from __future__ import division`. I'm pushing this fix now.

closed 2016-11-24 by ricardocrudo