KiCad/kicad-library-utils #63: Extra space in sch file

leventelist, 2016-10-24:
sch.py generates Fields with extra space at the output, for example:

-F 0 "R206" V 7680 6000 50 0000 C CNN
+F 0 "R207" V 7980 5800 50  0000 C CNN

Note the two white space after "50". This does nothing wrong except that eeschema puts only one space there, and all the time I run my script I have a ton of diffs where only the extra whitespace is the difference.

Thanks,
Levente

leventelist, 2016-10-24:
Okay Github optimizes out the two spaces after 50, but believe me that there are two spaces.

ricardocrudo, 2016-11-19:
Hello @leventelist which script exactly you are using? I guess add_part_number.py or update_footprints.py. It would be very handy if you could provide the files to reproduce the issue.

leventelist, 2016-11-19:
I use my own script that uses the sch.py from the kicad-library-utils repo.

You can find my script here:

https://github.com/leventelist/kicad-library-utils/blob/master/sch/update_db_footprints.py

I forked the repo.

To reproduce, you'd need a database, but I guess the update_footprint.py works the same. You can find some sample database here:

http://levente.logonex.eu/cuccok/device_db/db.sqlite

All you need to do is attach an attribute (sorry, I'm a gEDA refugee) to a component on the schematic, and run the script. Value, and the footprint will be changed.

In addition, I realized, that it is KiCad that add extra space, and the script adds just one space. Anyways, it produces extra diffs.

Please note that I am willing to add my database system to KiCad, if there's any interest.

Thanks,
Levente

ricardocrudo, 2016-11-22:
I remember this space issue was fixed already on KiCad side. Do you have a recent version?
In the PR https://github.com/KiCad/kicad-library/pull/464 @diggit says about have the build 6631 or higher.

leventelist, 2016-11-23:
This schematic was saved by KiCad 4.0.4. I don't know what does 6631 to version numbers.

http://levente.logonex.eu/cuccok/kicad_net/cpu.sch

HTH,
Lev

ricardocrudo, 2016-12-28:
I can confirm the extra space is added by KiCad. Because this is a kicad issue I'm closing the issue.

closed 2016-12-28 by ricardocrudo