KiCad/kicad-library-utils #64: New Rule: Disallow F.Silks for pads

pointhi, 2016-11-05:
https://github.com/KiCad/kicad-library/issues/774

CarlPoirier, 2016-11-05:
This is KLC 6.5!

pointhi, 2016-11-15:
yeah, I mean it would be a good idea adding a check into the ```check_kicad_mod``` script

ricardocrudo, 2016-11-19:
@CarlPoirier I don't think @pointhi's suggestion is actually covered by KLC 6.5. He is suggesting to not allow to have the F.Silks layer enabled at the Technical Layers. I guess we would need a new rule or an extra check (as we have for symbols). Having it we can extend the checking for the other layers as B.SilkS and maybe ***.User layers.

SchrodingersGat, 2017-01-12:
Hi @pointhi I have included this in the proposed update to KLC being dicussed here: https://github.com/KiCad/kicad-library/issues/879

Current draft: https://github.com/SchrodingersGat/kicad-library/wiki/KLC

Refer to Rule 9.3

SchrodingersGat, 2017-02-26:
@pointhi I am closing this as it has been addressed in the upcoming KLC

closed 2017-02-26 by SchrodingersGat