KiCad/kicad-library-utils #70: checklib.py is not running when called from comparelibs.py

Marsfan, 2016-12-13:
The title pretty much sums it up. I don't think I am calling anything wrong. I need this feature to work to get some continuous integration code I am using to work.

SchrodingersGat, 2016-12-13:
Can you provide any more info?

Marsfan, 2016-12-14:
@SchrodingersGat I called the function on bash as python3 comparelibs.py --check --new [path-to-new] --original [path to original]. The call properly returned the added/changed/removed parts, but instead of checking the added/changed parts, it simply returned the name of the part and the library directory.

Here is the execution: 
```
USER@USER_COMPUTER:/mnt/c/Users/USER/kicad-library-utils/schlib$ python3 comparelibs.py --check --original /mnt/c/Users/
USER/kicad-library/library/switches.lib --new /mnt/c/Program\ Files/KiCad/share/kicad/kicad-library/library/switches.lib

+ testingapart
USER@USER_COMPUTER:/mnt/c/Users/USER/kicad-library-utils/schlib$
```
I ran this on bash for Ubuntu for windows. I do not have access to a Linux machine and I am not good with the windows command line.

SchrodingersGat, 2016-12-28:
Ok I've found the error and submitted a PR here: https://github.com/KiCad/kicad-library-utils/pull/75

Thanks for finding this.

Marsfan, 2016-12-28:
@SchrodingersGat Thanks. As soon as this PR is merged I will get back to working on my automation script to identify errors in submitted parts.

SchrodingersGat, 2016-12-28:
Great, keep us posted :) I look forward to seeing what you come up with, and am happy to provide any assistance.

SchrodingersGat, 2017-02-13:
#75 has merged - you're good to go

Marsfan, 2017-02-13:
Booyah. Auto checking, here I come

closed 2017-02-26 by SchrodingersGat