KiCad/kicad-library-utils #82: Rule 10.5

WhiskyDelta, 2017-02-13:
Rule 10.5 says "Reference designator has a height of 2.0mm or smaller if needed and is placed on the fabrication layer."
check_kicad_mod.py reports an error if the reference is on the fabrication layer and reports no error if the reference is on the silk screen layer.
This should be corrected so that no error is reported if the reference designator is on the fabrication layer.

SchrodingersGat, 2017-02-13:
@WhiskyDelta thanks for reporting this. 

The entire KLC is currently being overhauled (behind the scenes) and will have a major update shortly.

Rest assured that this issue has been fixed in the updated KLC :)

poeschlr, 2017-08-23:
I think this has been corrected.

closed 2017-08-23 by poeschlr