KiCad/kicad-library-utils #89: courtyard rule is confusing...

SchrodingersGat, 2017-02-28:
@jkriege2 can you shed some light on this:

```
checking module: QFN-16-1EP_5x5mm_Pitch0.8mm
  Violating Rule 6.6
    Courtyard line has a width 0.05mm. This line is placed so that its clearance is measured from its
    center to the edges of pads and body, and its position is rounded on a grid of 0.01mm.
      A courtyard clearance in the range 0.2...0.3mm was found, but 0.25mm was expected.
      The recommendet courtyard rectangle would be:
          {'x': -2.95, 'y': -2.95, 'width': 5.9, 'height': 5.9}mm.
```

The rule recommends corners at 2.95, corners are at 3.00 so the courtyard is at least enough.

Plus, 0.25mm IS in the range 0.2...0.3mm...

What's going on here?

SchrodingersGat, 2017-02-28:
Looks like this might be addressed here - https://github.com/KiCad/kicad-library-utils/pull/88#issuecomment-282964934 ?

jkriege2, 2017-02-28:
The function currently works like this:
It calculates the smallest + largest courtyard clearance (which works, after the fix you referenced, in most cases now, I hope). That is the range given (maybe the text can be made a bit clearer here). What we currently expect is a clearance of exactly 0.15/0.25/0.5/...mm, so for this there is an error, because some clearance is 0.2mm and some is 0.3mm ... i.e. both do not match 0.25mm ...

I'm not really sure how to work with this rule ... some ideas come to mind:
1. We expect a clearance of AT LEAST XXmm but allow for larger clearances ... but then we don't give an error for e.g. 0.25mm clearance on a small SMD-part, where we expect 0.15mm
2. Currently the fix of this rule is optional (option `--fixmore`), but we could also make the check optional (but I don't like that option)
3. We leave it as it is ... but maybe need a better text ... we could e.g. give the list of all found clearances (4 values as this check is only performed for rectangular courtyards and the component is approximated by its bounding rect)

What do you think?
JAN

SchrodingersGat, 2017-02-28:
I think we should:

1. Specify minimum clearance e.g. 0.1mm
1. Specify other exceptions for e.g. connectors (0.5mm)
1. Check that the courtyard lines exist
1. Check that courtyard clearance is at least minimum (0.10mm)

The rest I think is human interaction. Otherwise there are too many edge cases!

I think the fix option should be:

`--add-couryard 0.2` adds an 0.20mm courtyard (simple rectangle)

Any more complicated fix should be left to the user

jkriege2, 2017-02-28:
I would like to keep the exact check and at least HINT the user that he might do better ... and how. Especially the last bit is convenient, I think, as hand-drawing the courtyard is often tedious and then wrong ...

eldarkg, 2017-03-18:
This check give failing check my current Pull Requests to module repos

closed 2017-10-05 by SchrodingersGat