KiCad/kicad-library-utils #94: Reference in Fabrication layer is missing

eldarkg, 2017-03-04:
I don't know is it right place for this issue.
Each module should contain reference in fabrication layer:
text field with name '%R'.
This is need for plotting assembly drawings.

jkriege2, 2017-03-04:
This will be part of the updated KLC: https://github.com/SchrodingersGat/kicad-library/wiki/KLC ... and the scripts can already add it with `--fix`. Some footprints also already have it, most not ... help on this is always welcom!

Best,
JAN

SchrodingersGat, 2017-03-04:
Please submit a PR for any footprints you spot without the extra RefDes

eldarkg, 2017-03-05:
@SchrodingersGat Are these FP don't generated with scripts?
For info: 
Pin_Headers.pretty: 845 entries.
TO_SOT_Packages_SMD.pretty: 74 entries.
Housings_DFN_QFN.pretty: 82 entries.
...

Manual add Ref field it is suicide.

jkriege2, 2017-03-05:
yes many are ... but I think the simples way is to use the `--fix` option of our check-scripts ...

jkriege2, 2017-03-05:
Also note that several repos are not up to speed with respect to the KLCs in diverse points ... but overhaul is (slowly) in progress!

JAN

SchrodingersGat, 2017-03-05:
Yes most of those are script generated - see https://github.com/pointhi/kicad-footprint-generator

eldarkg, 2017-03-05:
@SchrodingersGat Who will be regenerate footprints?

closed 2017-05-04 by eldarkg