KiCad/kicad-library-utils #95: Python version for `check_kicad_mod.py`

hackscribble, 2017-03-19:
I'm testing a footprint with `check_kicad_mod.py`. Using Python 2.7, it reports a courtyard error.  Using Python 3.5, it reports no errors.

With which version(s) of Python is `check_kicad_mod.py` intended to be used?

Thanks

SchrodingersGat, 2017-03-19:
Hi @hackscribble that's strange, I use Python3.6 and have no issues.

can you post the exact output you get for each python version, and the exact command you are running in each case?

hackscribble, 2017-03-19:
Here's what I get (running on Ubuntu 16.10).  And I have also listed the footprint file I'm checking (it's as per the KiCad repo but with a fix to the 3D model path).

```
ray@ray-Inspiron-7537:~/KiCad Contributing/kicad-library/modules/TO_SOT_Packages_SMD.pretty$ python --version
Python 2.7.12+

ray@ray-Inspiron-7537:~/KiCad Contributing/kicad-library/modules/TO_SOT_Packages_SMD.pretty$ python3 --version
Python 3.5.2+

ray@ray-Inspiron-7537:~/KiCad Contributing/kicad-library/modules/TO_SOT_Packages_SMD.pretty$ python /home/ray/Programs/kicad-library-utils/pcb/check_kicad_mod.py TO-252-2Lead.kicad_mod -v
checking module: TO-252-2Lead
  Violating Rule 6.6
    Courtyard line has a width 0.05mm. This line is placed so that its clearance is measured from its
    center to the edges of pads and body, and its position is rounded on a grid of 0.01mm.
      A courtyard clearance in the range -0.26...0.76mm was found, but 0.25mm was expected.
      The recommendet courtyard rectangle would be:
          {'y': -4.25, 'x': -5.7, 'height': 7.86, 'width': 13.66}mm.

ray@ray-Inspiron-7537:~/KiCad Contributing/kicad-library/modules/TO_SOT_Packages_SMD.pretty$ python3 /home/ray/Programs/kicad-library-utils/pcb/check_kicad_mod.py TO-252-2Lead.kicad_mod -v
checking module: TO-252-2Lead
  No violations found
```

```
(module TO-252-2Lead (layer F.Cu) (tedit 58CE4E80)
  (descr TO-252-2Lead)
  (tags "TO-252 2Lead")
  (attr smd)
  (fp_text reference REF** (at 0.3 -4.54) (layer F.SilkS)
    (effects (font (size 1 1) (thickness 0.15)))
  )
  (fp_text value TO-252-2Lead (at 0.25 4.81) (layer F.Fab)
    (effects (font (size 1 1) (thickness 0.15)))
  )
  (fp_text user %R (at 0.825 0.01) (layer F.Fab)
    (effects (font (size 1 1) (thickness 0.15)))
  )
  (fp_line (start -1.4 -3.04) (end -1.4 -3.39) (layer F.SilkS) (width 0.12))
  (fp_line (start -1.4 -3.39) (end 0.1 -3.39) (layer F.SilkS) (width 0.12))
  (fp_line (start -1.4 1.61) (end -1.4 -1.59) (layer F.SilkS) (width 0.12))
  (fp_line (start 0.1 3.41) (end -1.4 3.41) (layer F.SilkS) (width 0.12))
  (fp_line (start -1.4 3.41) (end -1.4 3.01) (layer F.SilkS) (width 0.12))
  (fp_line (start 4.68 -2.7) (end 5.88 -2.7) (layer F.Fab) (width 0.1))
  (fp_line (start 5.88 -2.7) (end 5.88 2.72) (layer F.Fab) (width 0.1))
  (fp_line (start 5.88 2.72) (end 4.67 2.72) (layer F.Fab) (width 0.1))
  (fp_line (start -1.32 1.74) (end -4.23 1.74) (layer F.Fab) (width 0.1))
  (fp_line (start -4.23 2.88) (end -1.32 2.88) (layer F.Fab) (width 0.1))
  (fp_line (start -4.23 1.74) (end -4.23 2.88) (layer F.Fab) (width 0.1))
  (fp_line (start -4.23 -2.86) (end -4.23 -1.72) (layer F.Fab) (width 0.1))
  (fp_line (start -4.23 -1.72) (end -1.32 -1.72) (layer F.Fab) (width 0.1))
  (fp_line (start -1.32 -2.86) (end -4.23 -2.86) (layer F.Fab) (width 0.1))
  (fp_line (start -1.32 3.36) (end 4.67 3.36) (layer F.Fab) (width 0.1))
  (fp_line (start 4.67 3.36) (end 4.67 -3.34) (layer F.Fab) (width 0.1))
  (fp_line (start 4.67 -3.34) (end -1.32 -3.34) (layer F.Fab) (width 0.1))
  (fp_line (start -1.32 3.36) (end -1.32 -3.34) (layer F.Fab) (width 0.1))
  (fp_line (start 7.45 -3.74) (end 7.45 3.76) (layer F.CrtYd) (width 0.05))
  (fp_line (start 7.45 -3.74) (end -5.7 -3.74) (layer F.CrtYd) (width 0.05))
  (fp_line (start -5.7 3.76) (end 7.45 3.76) (layer F.CrtYd) (width 0.05))
  (fp_line (start -5.7 3.76) (end -5.7 -3.74) (layer F.CrtYd) (width 0.05))
  (pad 1 smd rect (at -3.7 -2.29 270) (size 2 3.5) (layers F.Cu F.Paste F.Mask))
  (pad 3 smd rect (at -3.7 2.29 270) (size 2 3.5) (layers F.Cu F.Paste F.Mask))
  (pad 2 smd rect (at 3.7 0.01 270) (size 7 7) (layers F.Cu F.Paste F.Mask))
  (model TO_SOT_Packages_SMD.3dshapes/TO-252-2Lead.wrl
    (at (xyz -0.1377952756 0 0))
    (scale (xyz 1 1 1))
    (rotate (xyz 0 0 90))
  )
)
```

SchrodingersGat, 2017-03-19:
Can you try running with -vv rather than -v ?

hackscribble, 2017-03-19:
Using `-vv` argument on `check_kicad_mod.py` gives same results as `-v`.

I've tried a few of the other footprints in the repo.  They give the same results with both Python versions (either no violations or the same violations, depending on the footprint).

SchrodingersGat, 2017-03-19:
@jkriege2 do you run Windows or *nix?

jkriege2, 2017-03-20:
Hi! I'm running Win7 with Python 3.4.4

SchrodingersGat, 2017-03-20:
@hackscribble This is a bug with rule 6.6 - I think due to the way that Python2 and Python3 perform division.

There is a new version of KLC coming soon (https://github.com/SchrodingersGat/kicad-library-utils/tree/klc_rearranging) which fixes this.

Thanks for pointing it out.

closed 2017-03-20 by SchrodingersGat