KiCad/kicad-library-utils #96: kicad_mod.py fails to save file after issue fixing.

4x5dm, 2017-04-01:
Error log from console:
```D:\MyProjects\kicad-library-utils\pcb>python check_kicad_mod.py -v --fixmore d:\MyProjects\kicad_libraries\custom_kicad_footprints.pretty\Connector_Mini-DIN_Female_6Pin_2rows.kicad_mod
To print using colors you have to install colorama. Try to install it using: "pip install colorama"
[Continuing using no color mode]

checking module: Connector_Mini-DIN_Female_6Pin_2rows
  Violating Rule 6.6
    Courtyard line has a width 0.05mm. This line is placed so that its clearance is measured from its
    center to the edges of pads and body, and its position is rounded on a grid of 0.01mm.
      A courtyard clearance in the range 0.25...1.0mm was found, but 0.5mm was expected.
      The recommendet courtyard rectangle would be:
          {'y': -5.05, 'x': -9.81, 'height': 14.05, 'width': 17.0}mm.
        REPLACING Courtyard with RECTANGLE with clearance 0.5mm on layer F.CrtYd
Traceback (most recent call last):
  File "check_kicad_mod.py", line 132, in <module>
    module.save()
  File "D:\MyProjects\kicad-library-utils\pcb\kicad_mod.py", line 829, in save
    f = open(filename, 'w', newline='\r\n')
TypeError: 'newline' is an invalid keyword argument for this function
```

I can fix this issue and create a pull request.

4x5dm, 2017-04-01:
I use Python 2.7.

SchrodingersGat, 2017-04-04:
Hi @4x5dm please download the scripts again - there have been some major changes since yesterday. 

This issue should be fixed so let me know how you go. Thanks.

4x5dm, 2017-04-08:
All works. Thanks!

closed 2017-04-08 by 4x5dm