KiCad/kicad-library-utils #98: Rule 6.4 violation with asymmetric SMD package

hackscribble, 2017-04-03:
I'm designing a footprint for a TO-252-5-11 package (see image below).  The footprint fails check_kicad_mod.py with a rule 6.4 violation.  

Using the --fix option and comparing the resulting file with my original, it looks like the program calculates footprint centre based on the position of the **centres** of the pads.  For this type of package, this doesn't seem to correctly interpret the KLC rule "Generally [the middle] is the centroid calculated with respect to the **device lead ends**."

Am I interpreting this correctly, and am I safe to ignore this error?

Thanks.

[image]

[image]

hackscribble, 2017-04-03:
Attaching my original footprint and the version after --fix.

[footprints.zip](https://github.com/KiCad/kicad-library-utils/files/889692/footprints.zip)

SchrodingersGat, 2017-04-03:
Hi @hackscribble please refer to the [brand new KLC 2.0](https://github.com/KiCad/kicad-library/wiki/Kicad-Library-Convention) - this rule is now **8.2**

In this case the center of the footprint should be set to the datasheet position not the center position of the pads, as the exception to 8.2 states :)

hackscribble, 2017-04-03:
Thanks @SchrodingersGat.

The datasheet does not specify an origin for pick and place, which leaves ...

> Footprint anchor should be placed in the middle of the footprint (IPC-7351). Generally this is the centroid calculated with respect to the device lead ends.

So I guess I am OK to ignore the rule violation error, since I have calculated the centre based on lead ends.

Is there a new version of check_kicad_mod.py in the pipeline which will have the new KLC numbering?

SchrodingersGat, 2017-04-03:
> Is there a new version of check_kicad_mod.py in the pipeline which will have the new KLC numbering?

Yes, it should be merged any day now :)

hackscribble, 2017-04-03:
Thanks, I'll look out for it and re-run the test.

hackscribble, 2017-04-03:
I've got a fix to the `padsBounds()` function in `kicad_mod.py` so that it calculates the correct minimum bounding rectangle for non-symmetric arrangements of pads.  Should I put through a PR now or wait until the new release is out?

jkriege2, 2017-04-03:
new scripts have been merged!

hackscribble, 2017-04-03:
Thanks, @jkriege2.  I have looked at the latest version and I think the problem is still there.  My rationale is this ...

The check for Rule 8.2 calls the `padsBounds()` function, which passes the coordinates of the centres of the pads to the bounding box method:

```
        for pad in pads:
            pos = pad['pos']
            bb.addPoint(pos['x'], pos['y'])
```
 
Where all pads are the same size, the minimum bounding rectangle (MBR) of the centres of the pads has the same centre as the MBR for the pads themselves:

[image]

However, if the pads are of different sizes, the two MBRs have different centres:

[image]

My understanding is that Rule 8.2 should be checked against the red MBR in the image (as a proxy for the where the ends of the leads are), not the green one.

If this is correct, then couldn't Rule 8.2 use `overpadsBounds()` instead, which seems to take account of pad size when calculating the MBR.

```
           # Add each "corner" of the pad (even for oval shapes)
            
            p1 = _rotatePoint({'x': -sx/2, 'y': -sy/2}, angle)
            p2 = _rotatePoint({'x': -sx/2, 'y': +sy/2}, angle)
            p3 = _rotatePoint({'x': +sx/2, 'y': +sy/2}, angle)
            p4 = _rotatePoint({'x': +sx/2, 'y': -sy/2}, angle)
            
            points = [p1, p2, p3, p4]
            
            for p in points:
                x = px + p['x']
                y = py + p['y']
                bb.addPoint(x,y)
```

hackscribble, 2017-04-05:
PR submitted.

hackscribble, 2017-04-07:
PR merged: https://github.com/KiCad/kicad-library-utils/pull/107

closed 2017-04-07 by hackscribble