KiCad/kicad-library-utils #99: Errors with new scripts

jkriege2, 2017-04-03:
Hi @SchrodingersGat !

Could you please have another look at the new script version and/or Travis-Integration? In some cases it seems to work
https://github.com/KiCad/TO_SOT_Packages_SMD.pretty/pull/41
but here:
https://travis-ci.org/KiCad/kicad-library/builds/218111516
https://travis-ci.org/KiCad/kicad-library/builds/218111557
...
are some strange error messages.

Best,
JAN

SchrodingersGat, 2017-04-04:
Ok those errors are fixed -> see https://travis-ci.org/KiCad/kicad-library/builds/218283976

However the .pretty scripts are all "failing" in that they report zero errors even if there are errors -> https://travis-ci.org/KiCad/TO_SOT_Packages_SMD.pretty/builds/218111123

I shall sort this one out now.

SchrodingersGat, 2017-04-04:
Ok, that is fixed too.

SchrodingersGat, 2017-04-04:
Please double check but I think this is OK to close

closed 2017-06-05 by jkriege2