KiCad/kicad-library-utils #104: Incorrect check for Vss pin in rule4_5.py

hackscribble, 2017-04-05:
Latest KLC rule 4.5.ii treats Vss as a negative power pin which, like ground pins, should be at the bottom of the symbol.  The module `rule4_5.py` treats Vss as a positive power pin.

```
    def checkPowerPins(self):
        PWR = ['^[ad]*v(aa|cc|dd|ss|bat|in)$']
    
        first = True
        
        for pin in self.component.pins:
            name = str(pin['name'].lower())
            for pwr in PWR:
                if re.search(pwr, name, flags=re.IGNORECASE) is not None:
                    # Pin orientation should be "down"
                    if not pin['direction'] == 'D':
                        if first:
                            first = False
                            self.warning("Power pins should be placed at top of symbol")
                        self.warningExtra(pinString(pin))
```

I can work on a PR to move Vss to `checkGroundPins()` if that helps?

SchrodingersGat, 2017-04-07:
@hackscribble good find, I'll look into this.

`V+` and `V-` are not included in this check because often they are input pins e.g. for differential amps etc. Adding these to this check raises too many false positives.

hackscribble, 2017-04-07:
I have updated the PR to remove V+ and V-.

SchrodingersGat, 2017-04-07:
Closed, merged

closed 2017-04-07 by SchrodingersGat