KiCad/kicad-library-utils #105: datasheet-field BLANK no longer checked

jkriege2, 2017-04-05:
The datasheet field of symbols seems to be no longer checked for emptines (KLC 4.9.iv)

Best,
JAN

SchrodingersGat, 2017-04-07:
Datasheet field is checked in 4.10 and seems to be working fine.

I just tested with `checklib.py ~\kicad\library\*.lib -r 4.10 -s -vv` and it gave a lot of missing datasheet errors.

hackscribble, 2017-04-07:
4.9 script includes a check for the Datasheet field being invisible but not for emptiness:

```
    def checkDatasheet(self):
        
        # Datasheet field must be invisible
        fail = False
        
        ds = self.component.fields[3]
        
        if self.checkVisibility(ds):
            self.error("Datasheet field must be INVISIBLE")
            fail = True
            
        return fail
```

> Datasheet field is checked in 4.10 and seems to be working fine.

4.10 checks the Description and Keywords fields.  So I thought the "Datasheet" field checked by 4.10 is the "Documentation File Name" in the same dialog box.

[image]

SchrodingersGat, 2017-04-07:
Ah you are right @hackscribble - the datasheet field is in the LIB file not the DCM file.

SchrodingersGat, 2017-04-07:
PR here - https://github.com/KiCad/kicad-library-utils/pull/108

closed 2017-06-05 by jkriege2