KiCad/kicad-library-utils #109: check_kicad_mod.py - Rule 9.6 (minimum annular ring) issue

4x5dm, 2017-04-08:
I get ```Violating Rule 9.6 Minimum annular ring``` error on attached footprint. The error is generated for circular pads with 2.0mm diameter and 1.0mm drill. If diameter is set to 1.99mm or to 2.01mm, no error is occurs.

Here is what I get in console:
```
D:\some_path>python check_kicad_mod.py -v  d:\some_path\Cliff_UT1x_SPDT_ON-ON_Switch.kicad_mod
To print using colors you have to install colorama. Try to install it using: "pip install colorama"
[Continuing using no color mode]

Checking footprint 'Cliff_UT1x_SPDT_ON-ON_Switch':
  Violating Rule 10.4
    3D model reference
    No 3D model provided
  Violating Rule 9.6
    Minimum annular ring
    Pad 2 annular ring (0mm) is below minimum (0.15mm)
    Pad 4 annular ring (0mm) is below minimum (0.15mm)
    Pad 4 annular ring (0mm) is below minimum (0.15mm)
```

[Cliff_UT1x_SPDT_ON-ON_Switch.zip](https://github.com/KiCad/kicad-library-utils/files/907538/Cliff_UT1x_SPDT_ON-ON_Switch.zip)

hackscribble, 2017-04-08:
The values of 2mm and 1mm are being treated as integers by the script.  So when calculating the ring width ...

```
ring = (pad_x - drill_x) / 2
```
... the script works out the top line as integer 1, divides this by integer 2 and gets 0.

The fix is to make the bottom line a float:

```
ring = (pad_x - drill_x) / 2.0
```
I'll raise a PR for the library team to review.

SchrodingersGat, 2017-04-10:
@4x5dm can you please re-test this now that the proposed fix has been merged, and close the PR if the problem is now fully solved :)

4x5dm, 2017-04-10:
All solved.

closed 2017-04-10 by 4x5dm