KiCad/kicad-library-utils #119: Rule 4.12: check script misses invalid footprint filters

hackscribble, 2017-05-26:
My symbol for the LPC1111FHN33-101 in `nxp_armmcu.ib` which was merged a few weeks ago has what I now believe is an invalid footprint filter (invalid because it includes the number of pins):

```
$FPLIST
 *QFN*32*
 *HVQFN33*
$ENDFPLIST
```
The symbol passes `checklib.py` with no errors or warnings.  

The code in `rule4_12.py` includes this regex:

```
if re.search('(SOIC|SOIJ|SIP|DIP|SO|SOT-\d+|SOT\d+|QFN|DFN|QFP|SOP|TO-\d+|VSO|PGA|BGA|LLC|LGA)-\d+[W-_\*\?$]+', filter, flags=re.IGNORECASE) is not None:
    self.warning("Footprint filter '{filter}' seems to contain pin-number, but should not!".format(filter=filter))
```

I'm not a regex expert but I think the current expression is missing the "32" because of the "*" in the filter.  

Maybe the expression could be changed to match any of those keywords (e.g. SOIC) followed by 0 or 1 characters followed by a number?

BTW I will submit a PR to fix the filter.

jkriege2, 2017-05-26:
Hi!

I think that this is a hard problem to solve without using human intelligence ... e.g. if you ask for `SOIC` then allow `?*` and then expect a non-number, you would also get errors for the correct FPFilter like `SOIC*3.9x8.7mm*Pitch1.27mm*` ... I don't really know how to express this problem in a REGEXP ... I think what we did then the expression aove was devised is get errors at least for cases that are obviously wrong (with very low chance of throwing an error when there is really none)...

... But if you have a godd idea ... go ahead!
Best,
JAN

hackscribble, 2017-05-26:
V good points - I see the challenge now :smile: 

Maybe there could be an additional test that searched for the number of pins anywhere in the filter string. But that would still increase the likelihood of false positives.

closed 2017-05-26 by hackscribble