KiCad/kicad-library-utils #127: recognizing "simple" devices

jkriege2, 2017-06-07:
Dear all (especially @SchrodingersGat @poeschlr @pointhi ),

I'm opening this issues to discuss a possible extension for the utils:
Currently we get a lot of errors for "simple" devices (R,C,CP,L,D,transistors, diodes, ...) for which we allow deviations from KLC (especially they don't have background-filled 10mil rectangles, pins may lie on 50mil grid ...). We could avoid these, if we can recognize such devices. Here are some ideas for heuristics (I think there is no definite way, unless we can somehow get a flag into the file-format, so heuristics is all we can get):
- 1-3 pins, maybe 4-pins (not sure ... transistors in SO-8 may have stacked pins, so actually counting pin-stacks might be the way)
- small size (say. max. size of a coil- or resistor-symbol x1.5-x2)
- pin-names invisible or besides the pin, not "inside" device-box
- special lib-names: `device.lib`, `transistors.lib`, `diodes.lib` (already `opto.lib` is mixed!)
- maybe a special tag/keyword `simple-device` as "definite" marker

One might make a decision on say matching 80% of such heuristic points ...

What do you think? If I get some input, I ight go ahead and try to get some first version coded with which we can experiment.

Best,
JAN

suzizecat, 2017-06-15:
Couldn't you use the density of the symbol ? (The number of graphic lines within a certain space) and add a keyword so the script could report an error if there isn't the keyword and suggest adding it when you got a certain density ?

Julien

bobc, 2017-06-26:
A lot of logic devices for example in 74xgxx.lib also violate several of those rules. It's difficult to create compact symbols otherwise.

I'm trying to create small gates for 74xx.lib, but running into same problem. Also it would be good to make the "power unit" compact, e.g. 
[image]