KiCad/kicad-library-utils #129: checklib.py verbose flag not implicit when checking component

PhirePhly, 2017-06-11:
When trying to use schlib/checklib.py to check a single component, it fails with a Traceback. Adding the -v verbose flag causes it to behave as expected and give me the DRC errors. This is confusing since the help text for checklib states: `-c COMPONENT, --component COMPONENT check only a specific component (implicitly verbose)`

Example output without and with -v flag:
```
kenneth@thor:~/src/kicad-library/library$ ../../kicad-library-utils/schlib/checklib.py -c ATMEGA168A-AU ./atmel.lib 
Traceback (most recent call last):
  File "../../kicad-library-utils/schlib/checklib.py", line 68, in <module>
    if (args.verbose>2):
TypeError: unorderable types: NoneType() > int()

kenneth@thor:~/src/kicad-library/library$ ../../kicad-library-utils/schlib/checklib.py -c ATMEGA168A-AU -v ./atmel.lib 
Checking symbol 'ATMEGA168A-AU':
  Violating Rule 4.1 - Pin placement
    Pins must be placed on regular grid, and be of correct length
    Pins not located on 100mil grid:
     - Pin ADC7 (22), @ (-900, -350)
     - Pin (PCINT8/ADC0)PC0 (23), @ (1000, 250)
     - Pin (PCINT9/ADC1)PC1 (24), @ (1000, 150)
     - Pin (PCINT10/ADC2)PC2 (25), @ (1000, 50)
     - Pin (PCINT11/ADC3)PC3 (26), @ (1000, -50)
     - Pin (PCINT12/SDA/ADC4)PC4 (27), @ (1000, -150)
     - Pin (PCINT13/SCL/ADC5)PC5 (28), @ (1000, -250)
     - Pin ADC6 (19), @ (-900, -250)
     - Pin (PCINT14/~RESET~)PC6 (29), @ (1000, -350)
  Violating Rule 4.5 - Pin orientation
    Wherever possible, pins should be arranged by function
    Ground and negative power pins should be placed at bottom of symbol
    Positive power pins should be placed at top of symbol
  Violating Rule 4.6 - Pin types
    Pin electrical type should match pin function
    Pin types should match pin function
  Violating Rule 4.8 - Field text size
    Field text uses a common size of 50mils.
     - Pin (PCINT19/OC2B/INT1)PD3 (1), text size 40, number size 40
     - Pin (PCINT20/XCK/T0)PD4 (2), text size 40, number size 40
     - Pin GND (3), text size 40, number size 40
     - Pin VCC (4), text size 40, number size 40
     - Pin GND (5), text size 40, number size 40
     - Pin VCC (6), text size 40, number size 40
     - Pin (PCINT6/XTAL1/TOSC1)PB6 (7), text size 40, number size 40
     - Pin (PCINT7/XTAL2/TOSC2)PB7 (8), text size 40, number size 40
     - Pin (PCINT21/OC0B/T1)PD5 (9), text size 40, number size 40
     - Pin (PCINT22/OC0A/AIN0)PD6 (10), text size 40, number size 40
     - Pin AREF (20), text size 40, number size 40
     - Pin (PCINT16/RXD)PD0 (30), text size 40, number size 40
     - Pin (PCINT23/AIN1)PD7 (11), text size 40, number size 40
     - Pin GND (21), text size 40, number size 40
     - Pin (PCINT17/TXD)PD1 (31), text size 40, number size 40
     - Pin (PCINT0/CLKO/ICP1)PB0 (12), text size 40, number size 40
     - Pin ADC7 (22), text size 40, number size 40
     - Pin (PCINT18/INT0)PD2 (32), text size 40, number size 40
     - Pin (PCINT1/OC1A)PB1 (13), text size 40, number size 40
     - Pin (PCINT8/ADC0)PC0 (23), text size 40, number size 40
     - Pin (PCINT2/OC1B/~SS~)PB2 (14), text size 40, number size 40
     - Pin (PCINT9/ADC1)PC1 (24), text size 40, number size 40
     - Pin (PCINT3/OC2A/MOSI)PB3 (15), text size 40, number size 40
     - Pin (PCINT10/ADC2)PC2 (25), text size 40, number size 40
     - Pin (PCINT4/MISO)PB4 (16), text size 40, number size 40
     - Pin (PCINT11/ADC3)PC3 (26), text size 40, number size 40
     - Pin (PCINT5/SCK)PB5 (17), text size 40, number size 40
     - Pin (PCINT12/SDA/ADC4)PC4 (27), text size 40, number size 40
     - Pin AVCC (18), text size 40, number size 40
     - Pin (PCINT13/SCL/ADC5)PC5 (28), text size 40, number size 40
     - Pin ADC6 (19), text size 40, number size 40
     - Pin (PCINT14/~RESET~)PC6 (29), text size 40, number size 40
```

SchrodingersGat, 2017-06-13:
@PhirePhly thanks for raising this. @jkriege2 I believe this was introduced in one of your recent patches - I have added a comment to the particular line. If you have a chance it would be good to fix this soon.

jkriege2, 2017-07-26:
Fix is submitted:
https://github.com/KiCad/kicad-library-utils/pull/140

Ratfink, 2018-03-09:
There's no crash now, but it's still not implicitly verbose when -c is passed, though the help message says it is.  What is the desired behavior here, verbose or not?

Ratfink, 2018-08-03:
Personally I think changing the help message to not say that -c is implicitly verbose makes sense.  After all, it's not hard to type -vc instead, and [explicit is better than implicit](https://www.python.org/dev/peps/pep-0020/#id3).  Would anyone else care to weigh in?