KiCad/kicad-library-utils #130: Pin stacking incorrectly identified as violating rule 4.3.

Marsfan, 2017-06-15:
According to KLC 4.3.vi when pin stacking, when pins are output, all of the invisible pins should be identified as passive, while the visible one should be input. However, checklib.py reports this as an error due to there being different types of pins.

jkriege2, 2017-06-15:
We merged a PR about 1-2 weeks ago, which should fix this and allow the scripts to recognize pad-stacks: https://github.com/KiCad/kicad-library-utils/pull/116

If it is still an issue even with this PR, please post an example, so I can check and debug!

Best,
JAN

Marsfan, 2017-06-15:
It knows there is a stacked pin setup, but it does not understand the exemption. See [this travis build I did today](https://travis-ci.org/KiCad/kicad-library/builds/243420225) for [this PR](https://github.com/KiCad/kicad-library/pull/1206).

jkriege2, 2017-06-16:
Ahhhh ... wrong explanation from my side: If the pins are INPUTs (NOT power-inputs) all pins in the stack have to have type INPUT The stacks with PASSIVE pins are only required for output, power_out to prevent ERC-errors due to several sources driving a signal and for POWER_INPUT, because hidden power-inputs have a special function in KiCAD (they are used in power-flags!!!).

Hope that helps ;-)
JAN

Marsfan, 2017-06-20:
@jkriege2 Well, error on my side too. Looking this over, that should be an output, not an input.

jkriege2, 2017-06-21:
So, can we close this issue? Or did I miss another problem ...

Best,
JAN

closed 2017-06-21 by Marsfan