KiCad/kicad-library-utils #145: Travis returning error on small text thickness

Shackmeister, 2017-08-17:
Hey Travis familiar people

over at this PR Travis apparently disagrees with some text thickness
https://github.com/KiCad/Inductors_SMD.pretty/pull/19

according to KLC the fab layer REFDES can have a text thickness down to 0.01 but looking here:
https://github.com/KiCad/kicad-library-utils/blob/master/pcb/rules/klc_constants.py
it looks like Travis only accepts down to 0.025.. should this simply be changed?

evanshultz, 2017-08-17:
@SchrodingersGat 
I submitted the PR to drop it to 0.01mm in the scripts and Oliver updated KLC. However, it may be that Travis needs to get updated scripts and I'm not sure how to do that. I don't know if Oliver is back from his vacation yet, eight. For now, you're doing to right thing to merge it anyway.

FYI, currently there's only one setting for text thickness. I think the scripts should be changed to have a silk thickness and a fab thickness since they certainly have different requirements. Should be pretty easy to do.

SchrodingersGat, 2017-09-18:
@Shackmeister did this get resolved?

closed 2017-09-18 by Shackmeister