KiCad/kicad-library-utils #146: KLC violations with different copper and mask and stencil designs

evanshultz, 2017-08-18:
At https://github.com/KiCad/LEDs.pretty/pull/39 I submitted/fixed footprints to use a stencil design that is quite different from the pad design. I do not know if this was tried before.

Then at https://github.com/KiCad/Housings_QFP.pretty/pull/19 Jan was quite right to ask if there's a way to reduce the number of KLC violations based on the current scripts from this footprint design.

This method does create a lot of errors but it generates footprints that match the datasheet and known good design practices. The scripts require a mask and paste layer to exist for each pad. This is unreasonable since it doesn't allow custom mask/stencil designs, but it was a reasonable place to start. I think we're at a place where we need to consider how to do it better.

Perhaps a better way would be to have the script accept either of two padstack styles:
1. A copper pad has mask and paste layers. This is the current behavior.
2. A number of pads with the same number, considered as a group, contains elements on copper, mask and paste layers? This would allow for the footprints which currently give us problems, where the stencil design is different than the copper/mask design.

I can still imagine errors with the second method, but they should be in a very small minority of footprints and eliminate the issues I linked above. Other ideas?

This may drive some changes to KLC, depending on what we find (like the stencil pads use the pin number or not), but let's first figure out how to construct and check footprints.

s-light, 2017-09-01:
currently you have to give non-copper layer pads NO pin numbers - otherwise the ERC will fail (unconnected)
so 2. would have to check for position?! - this could get difficult as in your linked cases..
worse if you add thermal things like in this example:
[TI QFN with ThermalPad with Vias - Page 44](http://www.ti.com/lit/ds/symlink/tlc5971.pdf#page=44&zoom=auto,-266,760)

closed 2017-09-06 by evanshultz

s-light, 2017-09-11:
@evanshultz how is it fixed? can you link/comment so if someone will find this knows how to handle it?

evanshultz, 2017-09-11:
https://github.com/KiCad/Housings_QFP.pretty/pull/24 is a recent PR where the center thermal pad was updated on the TQFP-100 footprint. This can be a reference. General info:
- There is one pad on F.Cu and F.Mask layers.
- This pad has a pad number.
- All stencil openings are on only F.Paste layer.
- These openings have no pad number.
- The design of the stencil openings varies from part to part.

Currently the script does not accept a thermal pad when built correctly as described above, but it will be updated when possible to eliminate bogus failures.