KiCad/kicad-library-utils #166: Test naming convention in the scripts

antoniovazquezblanco, 2017-11-07:
Naming can be at least partially checked with the scripts.

SchrodingersGat, 2017-11-07:
Ok - do you want to implement this? :)

antoniovazquezblanco, 2017-11-09:
I was having a look at the existing scripts and found something. Can somebody tell me about G1.1 allowing characters `#` and `~`? What is the reason for it? Any place where this is documented?

SchrodingersGat, 2017-11-09:
Power symbols start with the `#` character, which is in general disallowed. I believe that the `~` character is used to designate a symbol with an invisible name.

This won't be allowed when the new symbol library comes into place, but I do not know how they will deal with that, as the symbol name will be the actual name of the file.

Ratfink, 2018-03-09:
https://github.com/KiCad/kicad-library-utils/pull/169 made some improvements for footprints.  I can see a few places to improve the code there further (like storing the compiled regular expressions in the class rather than in each instance, so they only have to be compiled once).

It would be good to update the G1_1 check for symbols to keep the two as similar as possible.