KiCad/kicad-library-utils #178: Substitute sexpr for python sexpdata package

antoniovazquezblanco, 2017-11-20:
I believe that outsourcing S-Expression parsing is a good idea.

Should I try change the actual sexpr implementation for the standard sexpdata python package? Will this changes be welcomed?

poeschlr, 2017-11-20:
Is the sexpr implementation available in a standard python 3.x installation? (What is the lowest version number where it exists?)

antoniovazquezblanco, 2017-11-20:
Sorry. Not standard installation. It is a package:

https://pypi.python.org/pypi/sexpdata

antoniovazquezblanco, 2017-11-21:
I agree this is not highest priority but it has benefits. This was another idea I had while looking at the utils code.

Using that package will simplify the existing code and it may also have performance improvements. As an example sexpdata has it's own tests thus removing the need to have complicated bash scripts that test wether the current implementation is performing as expected.

Ratfink, 2018-03-09:
sexpdata has unit tests, yes, but it's in alpha and hasn't had a commit since 2014-03-08.  That's an odd combination: its API is not officially stable, but it hasn't been modified for four years so it isn't *un*stable in practice.  I don't like the lack of development though.  I'm leaning towards adding documentation and unit tests to our own S-expression code over using this third-party package that looks abandoned.