KiCad/kicad-library-utils #189: checklib.py: AttributeError: 'Rule' object has no attribute 'footprints_dir'

lavadew, 2017-11-28:
When throwing an library at checklib.py, following error will be raised:

Traceback (most recent call last):
  File "./checklib.py", line 124, in <module>
    rule.check()
  File "/Volumes/Daten/git-stuff/kicad-library-utils/schlib/rules/S5_1.py", line 59, in check
    if self.footprints_dir and os.path.exists(self.footprints_dir) and os.path.isdir(self.footprints_dir):
AttributeError: 'Rule' object has no attribute 'footprints_dir'

jkriege2, 2017-11-28:
Hi!

can you copy the complete output + maybe provide the lib that causes the error (including DCM)?
Thanks,
JAN

lavadew, 2017-11-28:
Checking symbol 'AD390JD':
  Violating S4.2
    Pins should be arranged by function
    Ground and negative power pins should be placed at bottom of symbol
  Violating S4.4
    Pin electrical type should match pin function
    Power pins should be of type POWER INPUT or POWER OUTPUT
  Violating S5.2
    Footprint filters should match all appropriate footprints
    No footprint filters defined
Checking symbol 'AD558JN':
  Violating S4.2
    Pins should be arranged by function
    Ground and negative power pins should be placed at bottom of symbol
  Violating S4.4
    Pin electrical type should match pin function
    Pin types should match pin function
Checking symbol 'AD558JP':
  Violating S4.2
    Pins should be arranged by function
    Ground and negative power pins should be placed at bottom of symbol
  Violating S4.4
    Pin electrical type should match pin function
    Pin types should match pin function
  Violating S5.2
    Footprint filters should match all appropriate footprints
    No footprint filters defined
Checking symbol 'AD5593R':
  Violating S3.1
    Origin is be centered on the middle of the symbol
    Symbol not centered on origin
  Violating S4.4
    Pin electrical type should match pin function
    Pin types should match pin function
Traceback (most recent call last):
  File "./checklib.py", line 124, in <module>
    rule.check()
  File "/Volumes/Daten/git-stuff/kicad-library-utils/schlib/rules/S5_1.py", line 59, in check
    if self.footprints_dir and os.path.exists(self.footprints_dir) and os.path.isdir(self.footprints_dir):
AttributeError: 'Rule' object has no attribute 'footprints_dir'

I think i messed up my footprint paths, as i wanted to change it to the new kicad-footprint stuff..
I honestly have no clue whatsoever.

SchrodingersGat, 2017-11-28:
@lavadew thanks for reporting - I have submitted a fix for the scripts

lavadew, 2017-11-28:
Tried it with your fix - now it works as intended! Thank you for your work! :)

closed 2017-11-28 by lavadew