KiCad/kicad-library-utils #191: checklib.py: Issue with recognizing wildcards in footprint filters

lavadew, 2017-11-28:
I wanted to add a footprint filter of the EP1210 component in the Altera library, and i used the following string for it: DIP-40*
After checking with checklib.py, it said:
**  Violating S5.2
    Footprint filters should match all appropriate footprints
    Minuses and underscores in footprint filter 'DIP_40*' should be escaped with '?' or '*'.**

Then i changed it to DIP_40_*, hoping that this would fix the issue.
But again i got an error:
**  Violating S5.2
    Footprint filters should match all appropriate footprints
    Minuses and underscores in footprint filter 'DIP_40_*' should be escaped with '?' or '*'.**

lavadew, 2017-11-28:
I got the solution, if there is a dash or an underscore in the footprint filter name, it raises this warning.
But the EP1210 was produced as a DIL 40 only, as far i can tell from the research on the internet.
I find it a bit strange, because the more you can specify a footprint filter, the better, or am i wrong?

poeschlr, 2017-11-28:
The pin count is filtered by a separate filter in kicad.
This is why we typically do not include the pin count. (Only exception if your symbol would require an excessive amount of NC pins to fully qualify. Then you can leave the NC pins out and use the pin count within the footprint filter.)

For symbols that are only connected to one footprint, simply set the footprint field accordingly. (In addition to the more general footprint filter.)

lavadew, 2017-11-28:
Okay, got it.
Thank you for explaining that. :)