KiCad/kicad-library-utils #241: bug: checklib.py -  Test does not start on component TC2050

dirk-bst, 2018-05-17:
Terminal:
`./checklib.py -c TC2050 -vv  test.lib`
=> nothing is displayed

It works well with other components. 

[TC2050.zip](https://github.com/KiCad/kicad-library-utils/files/2012546/TC2050.zip)

Ratfink, 2018-05-23:
That's because the symbol is named `~TC2050` in that library.  See line 6 of the library file, which reads:

    DEF ~TC2050 J 0 40 Y Y 1 F N

Running `./checklib.py -c ~TC2050 -vv test.lib` works just fine.  I'm closing this because as far as I can tell there's no problem with the checking scripts.

How the file ended up this way, with the symbol name having a ~ in front of it but the documentation entry not having one, I don't know.  If it was only edited in KiCad, and you can give reproducible instructions to make that happen, perhaps you could report a bug over at https://bugs.launchpad.net/kicad so the KiCad developers can fix it.

closed 2018-05-23 by Ratfink

dirk-bst, 2018-05-24:
Hi,

thank you very much. I have no explanation for  `~`. But it was an library import from eagle. Maybe that can the reason, that it was broken.

Best regards