KiCad/kicad-library-utils #245: Improper handling of custom mask/paste pads

evanshultz, 2018-06-29:
We are using custom mask and paste pads as needed. Right now, this often generates a lot of noise because the footprint is constructed as desired but warnings and/or errors are generated.

See https://github.com/KiCad/kicad-footprints/pull/679 for one example. Here is the output from one footprint:
[image]

poeschlr, 2019-08-19:
One simple way out would be to ignore this single rule if the footprint name has the suffix ThermalVias. Even better would be an additional check if the pad number matches a smd pad. Would this be acceptable as i think this would be easy to implement.

poeschlr, 2019-08-19:
Ok there already is a pull request for adding a proper fix: https://github.com/KiCad/kicad-library-utils/pull/288