KiCad/kicad-library-utils #254: Silk, pad overlap seems to be unreliable

poeschlr, 2018-08-20:
I did not yet find time to investigate this in more detail. It reported an overlap between silk an pad in this PR even thought the effected pads have clearance between silk and pads.
https://github.com/KiCad/kicad-footprints/pull/839

zgyarmati, 2019-04-26:
This one also hit me, but from a different angle, on my footprint the script didn't report the overlap between the silkscreen and the pad, see here: https://github.com/KiCad/kicad-footprints/pull/1558#issuecomment-486897238
See this version of the footprint in question:
https://github.com/KiCad/kicad-footprints/blob/e52d325004eb2c6c4035ef20e5efcefed5f1e4a9/Connector_BarrelJack.pretty/BarrelJack_Lumberg_1613-14_15_SMT_Horizontal.kicad_mod

I started to look into the 
pcb/rules/F5_1.py' checkIntersections method, but before diving into it too deep, i need to ask:
Would it be possible to depend on external libs in the scriptset? For the intersection checking (but i think also at other places in the scripts) it would be handy to use shapely ( https://pypi.org/project/Shapely/ ) or other geometric lib.

evanshultz, 2019-04-30:
I think, at this time, we'd prefer to limit dependencies. Do you have some suggestions without adding any? If you feel the checking for this rule, at least, cannot be substantially or easily improved without adding an external lib then that is good info to share with us as well.