KiCad/kicad-library-utils #279: Mouser ECAD library and utils

cederom, 2019-01-09:
I just got an email from Mouser.. I have no experience with that but.. have you seen this free library of components? Would that apply and help in KiCAD? :-)

https://pl.mouser.com/electronic-cad-symbols-models/
http://mouser.componentsearchengine.com/pcb-libraries.php

poeschlr, 2019-05-31:
Free does not necessarily mean it is under a compatible license. If mouser wants to have their stuff in the official lib then they will need to open pull requests like everyone else and need to sign the license agreement. They would also need to follow the KLC like everyone else. And need to go through our rigorous vetting process. Meaning i am not sure what you want to tell us. Especially as you made this comment on the repo that holds the check scripts used for the official library instead of the library itself.

bobc, 2019-06-15:
The Mouser "free as in beer" data is provided by Samacsys, who are a commercial provider of data, so not "free as in speech".

No use at all to KiCad project. I think this issue can be closed.

poeschlr, 2019-06-15:
It is good news that kicad is supported but of no consequence for the official library. If you want to inform the community about this service then this is not the correct place to do so. The forum, reddit or any other platform would be better suited for this.

closed 2019-06-15 by poeschlr