KiCad/kicad-library-utils #291: Travis checks for packages3D

Shackmeister, 2019-03-17:
Hi Everybody

packages3D could definitely benefit from the Travis integration, which has been implemented.
Things which could be worth checking (open for ideas): 

1. If a pair of Step and wrl files has been comitted.

2. If KSU was used for exporting

3. If the internal stepname is the same as the external name.

4. if there is only one object inside the file.

5. A mechanical/solid check (might still be too unstable)

6.  Material properties.

7. if the model if referenced anywhere in the footprint repo.

I havent played with any Travis scripting, but thought I would put it here to discuss it.
@Misca1234 @easyw

Misca1234, 2019-04-06:
I am all for it
All tedious checking that is easy to implement as scripts is always good

I guess Travis for KiCad on git is implements as python scripts, like in
https://github.com/KiCad/kicad-library-utils

Perhaps we can place them in 
https://github.com/KiCad/kicad-library-utils

and super admins add them to the check in process flow

poeschlr, 2019-04-07:
Everyone with write permission should be able to setup travis.

Shackmeister, 2019-05-05:
Hi Guys

I made a (slightly crude) python script to check if KSU was used to export, and if the internal name is the same as the filename (very common issue).

This is by no means ready for anything, but can hopefully spark some discussion.
Also I'm a bit uncertain if we should base the scripts around FC/OCC/PYOCC or a handmade parser (what I used)

[Step_Check.zip](https://github.com/KiCad/kicad-library-utils/files/3146150/Step_Check.zip)

easyw, 2019-05-06:
@Shackmeister
checking if the exporting is done by KSU and if the file name is the same of the part name could be checked at simple text level. For KSU this check can be done also on WRL, at text level, adding also a check if wrl materials are used.
I think instead that we need to relay on FC or OCC or PyOCC to check if the part is a single object after importing the step file.
I wouldn't trust too much, finally, on the geometry check... it could be done as an option.

easyw, 2019-05-06:
@Shackmeister 
I've gone through the Step_Check.zip and I've found that you can add the check in OR for:
`/* author */ ('kicad StepUp','ksu'),` -> script generated file (already covered)
`('kicad StepUp'),('ksu MCAD'),` -> manually exported file (to be implemented)
for WRL files it should be enough to check for:
`#kicad StepUp wrl exported`
and to check if Materials are used, a check to:
`appearance Appearance {material DEF` could be implemented.