KiCad/kicad-library-utils #292: Error following modification of check scripts

myfreescalewebpage, 2019-03-24:
Following https://github.com/KiCad/kicad-library-utils/pull/274, I have the result shown below on https://github.com/KiCad/kicad-symbols/pull/1601:

[image]

@poeschlr @fauxpark please have a look, thanks. Joel

fauxpark, 2019-03-24:
Odd, Travis is still using Python 2...

fauxpark, 2019-03-25:
Since build #5023 Travis has been using Python 3 with no complaints.

fauxpark, 2019-03-25:
@myfreescalewebpage this build/PR is quite old, but only ran recently, evidently in the period between #274 being merged and the update to Python 3. Did you re-run this build to test it out?

myfreescalewebpage, 2019-03-26:
@fauxpark it was the last time run 2 days ago. The PR is now merged and we are now more able to restart the run (it fails because PR is merged).

myfreescalewebpage, 2019-03-26:
Seems not to be present on the last run in the pull request I'm currently reviewing. Closing the issue at the moment.

fauxpark, 2019-03-26:
I think this is actually expected behaviour if Travis pulls the .travis.yml from the same point in the Git history as the PR. @poeschlr advised not to bother with support for Python 2 in this rule, or the check scripts in general.

So, it doesn't fail because my PR was merged, it fails because it's trying to use Python 2 to check this rule.

myfreescalewebpage, 2019-03-30:
@fauxpark seen again today 2019-30-03, reopening the issue.

myfreescalewebpage, 2019-03-30:
@poeschlr I'm not sure but it seems that:
- travis is triggered with python 3 when a commit is pushed
- travis is triggered with python 2 when we relaunch the tests manually with the "Restart build" button....

fauxpark, 2019-03-30:
That is interesting. Evidently Travis pulls the .travis.yml from current master for a new build, but when rerunning an existing build the Python version is already stored somewhere in the metadata for the build so it simply uses that one.

@myfreescalewebpage this thread appears to offer a solution: https://travis-ci.community/t/can-restart-build-to-use-latest-travis-yaml-configuration/1384

myfreescalewebpage, 2019-04-01:
@fauxpark the main issue is that it is not "my" pull request, this is the pull requests of people contributing to the library that we are reviewing... Keeping compatibility with python2 should be very great, isn't it possible ?

fauxpark, 2019-04-01:
It is possible, in fact that's what my PR did originally, however Rene seemed to be saying not to bother: https://github.com/KiCad/kicad-library-utils/pull/274#discussion_r253264628

So I defer to him as to what to do next.

poeschlr, 2019-04-01:
Yes travis is stuck at python 2 until you close and reopen the pull request. I know this is annoying especially as we now kind of rely on python 3 for the tests. (I did not know about this travis restriction when i merged the changes. Otherwise i would have updated to python 3 a bit before merging the changes that require python 3.)

---

And no python 2 is dead. I am not prepared to invest any time in getting any script compatible with both.

myfreescalewebpage, 2019-04-02:
@poeschlr is it tested that closing/opening the PR will restart travis with the new configuration, i.e. running python3 ?

myfreescalewebpage, 2019-04-04:
Just confirmed, the issue can be solved by closing/opening the pull request. Closing this issue.

closed 2019-04-04 by myfreescalewebpage