KiCad/kicad-library-utils #300: kicad-library-utils don't work

jneiva08, 2019-05-24:
Hi

I'm trying to run check_kicad_mod.py but it don't work.

Its giving the following error:
```
Traceback (most recent call last):
  File "./check_kicad_mod.py", line 117, in <module>
    rule.check()
  File "/home/jneiva/workspace/eda/kicad-library-utils/pcb/rules/G1_7.py", line 17, in check
    if 'linux' in platform.platform().lower() and not checkLineEndings(self.module.filename):
  File "/home/jneiva/workspace/eda/kicad-library-utils/common/rulebase.py", line 75, in checkLineEndings
    LE1 = ord(chr(filecontentsraw[-2]))
TypeError: an integer is required
```
Have try checklib.py it don't works too.

Its giving the following error:
```
Traceback (most recent call last):
  File "./checklib.py", line 135, in <module>
    rule.check()
  File "/home/jneiva/workspace/eda/kicad-library-utils/schlib/rules/G1_7.py", line 19, in check
    if not checkLineEndings(self.component.lib_filename):
  File "/home/jneiva/workspace/eda/kicad-library-utils/common/rulebase.py", line 75, in checkLineEndings
    LE1 = ord(chr(filecontentsraw[-2]))
TypeError: an integer is required
```

My OS is Linux Mint xfce 19.1

Solartraveler, 2019-05-31:
Yes, just came across the same error with Debian Stretch.
As workaround, I simply disabled the checks:
    LE1 = 42 #ord(chr(filecontentsraw[-2]))
    LE2 = 23 #ord(chr(filecontentsraw[-1]))

The you can at least run the other checks.

poeschlr, 2019-05-31:
What python version are you using? We definetly require pyhton 3 and might even require at least 3.4.

jneiva08, 2019-06-01:
Have alwats run the script by "./checklib.py" and it always worked but now gives the error mantioned earlier.
but if i run like you say "python3 checklib.py" it don't gives any error.

StormTrooper, 2019-06-14:
Had the same problem. I have python 2.7 and 3 intstalled
Running ./checklib.py on my desktop used python 2.7
I changed first line of checklib.py to:
[code]
#!/usr/bin/env python3
[/code]

evanshultz, 2019-06-14:
Python 3 is required. This is stated clearly above. Please use Python 3 to run the scripts and post here if and only if you still have problems. Thank you.

poeschlr, 2019-06-15:
Evan the first line of the script was indeed wrong. Fix is in pull request #305

evanshultz, 2019-06-16:
I misunderstood the comments above. Sorry.

PR #305 (2019-06-16): Ensure python 3 is used when running script directly
Fixes #300
closed 2019-06-16 by evanshultz