KiCad/kicad-library-utils #308: S4.3 pin stacking not reliable enough

karlp, 2019-07-03:
It isn't reliable enough to be considered a "failure" (yet?) so it's not looked at by submitters, "all checks are green"  And when you do read the suggestions, it just says, "power in/out should not be invisible"  It doesn't mention the other option, "should this pin be passive?"  

https://github.com/KiCad/kicad-library-utils/blob/master/schlib/rules/S4_3.py#L135 is correctly idenfitying the "power in/out+passive" stack that's required, when it's correct, but it doesn't suggest it as a solution when it's wrong.

poeschlr, 2019-08-19:
Was the invisible power input pin already stacked? Because that is the only case where we could make this suggestion. If it is not stacked then the error is that it is invisible. If it is stacked then the error is that it has potentially the wrong type.

karlp, 2019-08-19:
yes, because I'd attempted to get the stack right.