KiCad/kicad-library-utils #317: GitLab Library Utils migration

sethhillbrand, 2019-11-27:
Hi Librarians-

As part of the Great KiCad GitLab migration of 2019, I am testing the movement of open issues and PRs from GitHub to GitLab.

Here's what we know so far:

- Open Pull Requests will transfer to GitLab without moving the source of the PR.  That means pull requests will become GitLab Merge Requests that are referenced to the user's GitHub repository
- Open Issues will transfer to GitLab issues
- Closed/Resolved PRs and Issues will also transfer

In order to facilitate the transfer, it will be helpful if you can do two things:
- Open a GitLab account (these can use your GitHub OAuth)
- **Set your GitHub email to public**
- Let me know who the responsible parties are for the repository

The second point is critical because the migration script will only link users whose e-mail addresses match.  If the script can't see your GitHub e-mail address, the issue will be tied to the KiCad-Bot that performs the migration.

The timeline for the actual migration will be determined by the responsible parties from bullet 3 above but it would greatly assist my testing for bugs if you could do the first two steps above as soon as possible.

Thank you-
Seth

evanshultz, 2019-11-27:
@poeschlr 
The official symbol generator is rather buried in this repo at https://github.com/KiCad/kicad-library-utils/tree/master/schlib/autogen. Would the GitLab migration be a good time to promote this to it's own repo? Perhaps creating https://gitlab.com/kicad/libraries/symbol-generator? (A complimentary location could be for the footprint generator, if we are able to add it, and any 3D model generators if we wish to include them.)

poeschlr, 2019-11-28:
Would definitely be an option. I will also ask @pointhi  if it might be a good idea to put the footprint generator under the kicad group to integrate it better (and make it more likely that users find it).

pointhi, 2019-11-28:
@poeschlr: would be okay for me: https://github.com/KiCad/kicad-footprints/issues/1983#issuecomment-559396754

My suggestion would be putting those things into a sub-directory in library, like https://gitlab.com/kicad/libraries/utils or create a own https://gitlab.com/kicad/utils subgroup. For example: we should create docker images for the CI system, which need to be put somewhere as well.