KiCad/kicad-library-utils #328: Symbol library version control?

nacansino, 2020-06-08:
I am currently experimenting with `schlib/comparelibs.py` to integrate a version control system for my symbol libraries. Our custom library structure is of the following form:

```text
xlab
|   Resistor.dcm
|   Capacitor.dcm
|   ...
|
|   Resistor.lib
|   Capacitor.lib
|   ...
|
|   Footprint
|   |   Resistor.pretty
|   |   |   `RESXX123`.kicad_mod
|   |   |   ...  
|   |
|   |   Capacitor.pretty
|   |   |   `CXYZ456`.kicad_mod
|   |   |   ...  
|   |
|   |   ...
|   |
|
|   3D
|   |   Resistor.3dshapes
|   |   |   `RESXX123`.stp
|   |   |   ...  
|   |
|   |   Capacitor.3dshapes
|   |   |   `CXYZ456`.stp
|   |   |   ...  

```

The end goal is to see if there are additions, modifications, or deletions in my symbol libraries (`Resistor.lib`, `Capacitor.lib`, etc). Unlike the changes in `Footprint/` and `3d/` that are explicit as the components are separated per file, components inside the symbol libraries are lumped into groups.

While I was able to verify that `schlib/comparelibs.py` does the job well when comparing libraries, it requires that you have the whole library for both versions to work properly. This does not fit Git, because by the design, in only stores the deltas between changes, 

The only solution I'm thinking for full Git VCS integration is to create a script that can read the temporary deltas Git is creating when modifying the repository, then call that script as an external diff tool.

Has anyone dealt with this task before? What are your approaches?

eeintech, 2020-06-08:
@nacansino My current method to check on library part changes is manual as I use diff or meld to visually check on those.
However, I see where you're going and have some thoughts to share. It looks like the `comparelibs.py` is actually scanning all parts in the "new" lib file, not only the ones that were updated, maybe this is setup problem on my side. Anyway, what about using a diff tool to only export the updated or "new" parts from both older and newer rev into temporary/delta lib files then running the script on those library files instead?
What are you exactly trying to catch? More like KiCad rule breaking concerns or actual check on data integrity? Both?

evanshultz, 2020-06-11:
I think you will get more responses if you post this question at https://forum.kicad.info/. That forum is for KiCad users.