KiCad/kicad-library-utils #329: F5.3 and checking non-line elements

mibe, 2020-06-19:
I was wondering why the F5.3 check at KiCad/kicad-footprints#2267 failed. The problem here is the check expects the elements to be line elements only, while the footprint contains arc elements on the CrtYd layer. Which is against the KLC, I guess ("The [...] courtyard is defined as the [...] **rectangular area** [...]. It is allowed to create a contoured courtyard area using a **polygon** instead of a simple rectangle.")? On the other hand the check does not complain about having non-line elements on the CrtYd in the first place.

So what's your opinion about this? Possible solutions could be:
- ignoring non-line elements in the code and/or
- adding a check to enforce line elements only

evanshultz, 2020-06-19:
Some background info: Because of the ways arcs are represented internally, the endpoints are problematic and thus not allowed in KLC. Which you noted. Only one end of the arc has an absolute location so ensuring it's closed is... not easy.

cpresser, 2020-06-19:
I took a quick look.
Indeed, its a issue with the endpoint of the arc (numeric accuracy)

Getting the endpoint returns `-6499999` which is not on the grid.
`{'start': {'x': 0, 'y': 0}, 'end': {'x': -6.5, 'y': 8.5}, 'angle': -74.81070976, 'layer': 'F.CrtYd', 'width': 0.05} [6499999, 8500000, -6500000, 8500000]
{'start': {'x': 0, 'y': 0}, 'end': {'x': 6.5, 'y': -8.5}, 'angle': -74.81070976, 'layer': 'F.CrtYd', 'width': 0.05} [-6499999, -8500000, 6500000, -8500000]`

I will have a look at the function and see if its accuracy can be improved.

mibe, 2020-08-26:
@cpresser Did you have a chance to look if the accuracy can be improved?

cpresser, 2020-09-05:
Hi @mibe 
Thanks for the reminder. I forgot about this issue. But fortunately its an easy fix.
I have pushed a change of one line to #344
With that, the check passes with no errors:
[image]

mibe, 2020-09-05:
I just tried the change and can confirm that #334 fixes this issue.

cpresser, 2020-09-13:
@evanshultz would you please review/merge this one?

PR #334 (2020-09-16): Add rouding to mmToMicrons() function
For when using getStartPoint() on arcs combined with the mmToMicrons()
function the accuracy was not good leading to errors of the F5.3 check.
See https://github.com/KiCad/kicad-footprints/pull/2267

With rounding applied to the value before converting it to int that issue
is gone. This change won't break other rules since the function mmToMicrons()
is only used for rule F5.3.

Fixes #329
commit 41297f9e (2020-09-16): Add rouding to mmToMicrons() function

For when using getStartPoint() on arcs combined with the mmToMicrons()
function the accuracy was not good leading to errors of the F5.3 check.
See https://github.com/KiCad/kicad-footprints/pull/2267

With rounding applied to the value before converting it to int that issue
is gone. This change won't break other rules since the function mmToMicrons()
is only used for rule F5.3.

Fixes #329
closed 2020-09-16 by evanshultz