KiCad/kicad-library-utils #333: The list of datasheets is unfortunately somewhat old and therefore incomplete. Complete series like STM32G4 are missing.

Franck78, 2020-09-01:
The list of datasheets is unfortunately somewhat old and therefore incomplete. Complete series like STM32G4 are missing.
It is possible to extract a list of all STM32 datasheets from STs website with only a few lines of python code. PR coming soon...

_Originally posted by @rleh in https://github.com/KiCad/kicad-library-utils/pull/326#issuecomment-631271799_

Franck78, 2020-09-01:
Extracted with some search and replace + libreoffice calc.

The link are more complicated but working.
189 datasheet from MCU category

[mcu - datasheet - Resources - Search STMicrocontrollers.txt](https://github.com/KiCad/kicad-library-utils/files/5158220/mcu.-.datasheet.-.Resources.-.Search.STMicrocontrollers.txt)

rleh, 2020-09-01:
I created a script to retrieve all datasheet in #327:

https://github.com/KiCad/kicad-library-utils/blob/6bedf5f01133c1f6b379f9e0fc0e76fca9f6e924/schlib/autogen/stm32/stm32_get_datasheets.py

Franck78, 2020-09-01:
It was "coming soon" dated 20 may when I red #326 
I guess you updated it just now with 'done'.

closed 2020-09-01 by Franck78

rleh, 2020-09-01:
Unfortunately I forgot to update my comment https://github.com/KiCad/kicad-library-utils/pull/326#issuecomment-631271799 when I created #327 the same day. Updated it after I saw this issue to avoid confusion in the future.

But #327 is still not reviewed and merged...

Franck78, 2020-09-01:
my manual extract returns 189 datasheets

Your script returns ~140

You can update the static list (unless you are powerless user as me)

"still not reviewed and merged..." :  yes yes , it seems a lot of thing are pending here.

You have this waiting: https://github.com/KiCad/kicad-packages3D/pull/714

rleh, 2020-09-01:
> my manual extract returns 189 datasheets
> Your script returns ~140

Interesting. I'll investigate...

Franck78, 2020-09-01:
on st.com
* search datasheet. 6000 files
* refine by product
* refine by document type
189 files selected
and there is no G4

rleh, 2020-09-01:
My list is (intentionally) missing the STM32MP family (6 datasheets, see https://github.com/KiCad/kicad-library-utils/pull/327/files#diff-f6a681b02425513e05a08aa469e1ea28R21).

Your "STM32" list is longer because it contains datasheets of STM8 and other non-STM32 controllers like e.g. [ST92124](https://www.st.com/content/ccc/resource/technical/document/datasheet/20/9c/84/a1/13/f8/4c/7c/CD00170892.pdf/files/CD00170892.pdf/jcr:content/translations/en.CD00170892.pdf) or [ST7LITE1](https://www.st.com/content/ccc/resource/technical/document/datasheet/10/d7/bc/e9/56/96/41/ec/CD00003365.pdf/files/CD00003365.pdf/jcr:content/translations/en.CD00003365.pdf).

rleh, 2020-09-01:
To get the number (and a list) of existing _STM32_ datasheets:
* Go to https://www.st.com/en/microcontrollers-microprocessors/stm32-32-bit-arm-cortex-mcus.html
* For each STM32 family:
  * Click _Resources_ -> _Technical Literature_ -> _Datasheet_

Franck78, 2020-09-02:
oh ok. And the script cannot process datasheets not STM32, or output some drafts to begin a symbol ?

rleh, 2020-09-02:
No. `stm32_generator.py` only generates STM32 symbols.
The main data source for the symbol generation is [STM32CubeMX](https://www.st.com/en/development-tools/stm32cubemx.html) ([see README](https://github.com/KiCad/kicad-library-utils/blob/master/schlib/autogen/stm32/README.md)), I'm not aware of a similar data source for other controllers.

Franck78, 2020-09-02:
I don't know ST micro families. I you tell me 'this model' is worth being created, I'll do it. The main concern being avoid entering manually 100 pins with names&type.

-grab them with pdf extract
-grab them from .h includes dev files